RobotLocomotion/director #581: Build fails to find Qt4 on Ubuntu 16.04

rdeits, 2018-01-19:
Hi Pat! I'm running into a strange build issue, and I was hoping you could help. I currently can't build Director on my Ubuntu 16.04 system. Running the superbuild cmake gives:

```
CMake Error at /home/rdeits/tools/cmake-3.5.2-Linux-x86_64/share/cmake-3.5/Modules/FindPackageHandleStandardArgs.cmake:148 (message):
  Could NOT find Qt4 (missing: QT_QTOPENGL_INCLUDE_DIR QT_QTOPENGL_LIBRARY)
  (found version "4.8.7")
Call Stack (most recent call first):
  /home/rdeits/tools/cmake-3.5.2-Linux-x86_64/share/cmake-3.5/Modules/FindPackageHandleStandardArgs.cmake:388 (_FPHSA_FAILURE_MESSAGE)
  /home/rdeits/tools/cmake-3.5.2-Linux-x86_64/share/cmake-3.5/Modules/FindQt4.cmake:1333 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  /home/rdeits/locomotion/director/cmake/dd-macros.cmake:22 (find_package)
  cmake/externals.cmake:63 (setup_qt)
  CMakeLists.txt:27 (include)
```

I definitely have Qt4 installed (and have installed all of the `apt` packages from the Readme), but Director seems to be unhappy with that installation for some reason. We just tried building on Twan's computer (with 16.04) and it worked, so it's clearly something specific to my machine, but I don't even know where to begin. Any suggestions?

rdeits, 2018-01-19:
Aha! It looks like I just needed `libqt4-opengl-dev` as well. I guess we should add that to the Readme.

patmarion, 2018-01-19:
hmm, I wonder why the deps in the Readme were not sufficient.  Here is the dep install script that works for ubuntu 16 in docker, btw:

https://github.com/RobotLocomotion/director/blob/master/distro/travis/install_deps.sh

rdeits, 2018-01-19:
Yeah, I'm also confused about that. Seems to be some artifact of the way my desktop is configured. Perhaps `qt4-opengl` was removed when I had to clean out the `mesa-dev` stuff that an old Drake version installed.

closed 2018-01-19 by rdeits

manuelli, 2018-07-31:
I ran into this as well while building inside the docker image. Installing `libqt4-opengl-dev` fixed the issue for me as well. For reference the director options I am setting are shown below.

```
cmake $director_src_dir/distro/superbuild \
      -DUSE_EXTERNAL_INSTALL:BOOL=ON \
      -DUSE_DRAKE:BOOL=OFF \
      -DUSE_LCM:BOOL=ON \
      -DUSE_LIBBOT:BOOL=ON \
      -DUSE_SYSTEM_EIGEN:BOOL=ON \
      -DUSE_SYSTEM_LCM:BOOL=OFF \
      -DUSE_SYSTEM_LIBBOT:BOOL=OFF \
      -DUSE_SYSTEM_VTK:BOOL=OFF \
      -DUSE_PCL:BOOL=ON \
      -DUSE_APRILTAGS:BOOL=ON \
      -DUSE_KINECT:BOOL=ON \
      -DCMAKE_INSTALL_PREFIX:PATH=$install_dir \
      -DCMAKE_BUILD_TYPE:STRING=Release
```

ajeyaajeya, 2019-09-11:
I face the same error . I have libqt4-opengl-dev already installed .
I get the following error on running cmake ..
I am expecting a normal cmake without error.

```
ajeya@node1:~/gr-inspector/build$ cmake ..
-- Build type not specified: defaulting to release.
CMake Deprecation Warning at CMakeLists.txt:56 (cmake_policy):
  The OLD behavior for policy CMP0026 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   filesystem
--   system
--
-- Python checking for PyQt4
-- Python checking for PyQt4 - found
CMake Error at /usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find Qt4 (missing: QT_QWTPLOT3D_INCLUDE_DIR QT_QWTPLOT3D_LIBRARY)
  (found suitable version "4.8.7", minimum required is "4.2.0")
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.10/Modules/FindQt4.cmake:1327 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:152 (find_package)

-- Configuring incomplete, errors occurred!
See also "/users/ajeya/gr-inspector/build/CMakeFiles/CMakeOutput.log".
See also "/users/ajeya/gr-inspector/build/CMakeFiles/CMakeError.log".

```

@manuelli can you please point me to the "director options that you have listed above" . How do i find this file?