RobotLocomotion/director #590: when compiling Director from source.  After Cloning into 'vtk' ... did not continue.

QiayuanLiao, 2018-01-31:
```
Scanning dependencies of target vtk
[  2%] Creating directories for 'vtk'
[  4%] Performing download step (git clone) for 'vtk'
Cloning into 'vtk'...
```

patmarion, 2018-01-31:
Perhaps the VTK git server was temporarily offline?  Can you try it again, does it happen everytime?  Also note that the VTK git history could be large so the download might take a while depending on the network connection.

QiayuanLiao, 2018-02-01:
I am in China, the foreign network speed, but I use shadowsocks to accelerate.When I used the Precompiled Drake, the Cloning into 'vtk'... was very fast, I don't understand?

QiayuanLiao, 2018-02-01:
What files path the Compiler clone into?

patmarion, 2018-02-01:
The clone should work if the network permits it.  Is it eventually timing out and printing a relevant error message?

In the default build configuration on Ubuntu, the director build will download pre-compiled vtk libraries, it is not cloned.  Are you building on a mac?  On mac, by default, vtk is cloned and compiled from source.  You can configure with `USE_SYSTEM_VTK` in order to use vtk from homebrew, then you can avoid doing the clone from source.

QiayuanLiao, 2018-02-03:
Well after about 6 hours clone, I compiled successfully, can we add a progress bar about clone?

closed 2022-10-02 by qiayuanliao