RobotLocomotion/director #592: `TaskRunner` prevents Director from exiting

rdeits, 2018-02-02:
It looks like the `TaskRunner` itself is keeping the Director app alive even when the window is closed (or when ctrl+Q is pressed). To reproduce, launch the `drake-visualizer` app and then run (in the console):

```python
from director import taskrunner
import time
def task():
    while True:
        time.sleep(0.1)
tr = taskrunner.TaskRunner()
tr.callOnThread(task)
```

then close the Director console and main window. The `drake-visualizer` binary keeps running until you `ctrl+c` it from the terminal. 

I'm trying to figure out how to fix this, but haven't had any luck yet.

rdeits, 2018-02-02:
I can get around this by changing the loop to `while mainWindow.isVisible():`. Is that the right thing to do?

patmarion, 2018-02-02:
oh, i thought it was zeromq that was preventing the exit, but it looks like it is any python thread. hmm.

there is this signal you can catch:
```
def on_quit():
    print('quitting')
app.applicationInstance().connect('aboutToQuit()', on_quit)
```

if the user quits by closing the window, you can use the `isVisible()` trick, but if they press `control+Q` then the app quits before the window is closed and it still hangs.  I wasn't able to find anything that works except to modify `taskrunner.py` and make it a daemon thread:

```
  def callOnThread(self, func, *args, **kwargs):
    t = Thread(target=lambda: func(*args, **kwargs), daemon=True)
```

then it exits fine.  still not ideal.

rdeits, 2018-02-02:
Ok, makes sense. Making the thread a daemon solves all of my immediate problems, so I'll go with that for now.

patmarion, 2018-02-02:
Ok good.

I still don't understand why it doesn't work.  It seems like when the Qt event loop exits the Python thread loses it's ability to run.  Code on the main Python thread runs fine, but calling thread.join() just hangs.  I think it might have something to do with PythonQt deinitializing some python state.  I'll let you know if I am able to debug it further.