RobotLocomotion/director #605: AttributeError: 'module' object has no attribute 'viewer_load_robot_t'

AlameluThinnappan, 2018-04-06:


patmarion, 2018-04-06:
Is it possible that you have a conflicting version of libbot installed somewhere else?  Did you enable the USE_SYSTEM_LIBBOT variable when configuring director?

patmarion, 2018-04-08:
It looks like you might be configuring wrong.  Please see the toplevel README for instructions.  It should look something like this:

```
cd director
mkdir build && cd build
cmake ../distro/superbuild
make
```

If you want, you can add `-DUSE_LCM=ON`.  You might also consider using precompiled binaries here:  https://bintray.com/patmarion/director/director/0.1.0-266-g071a233#files

Or possibly use another project like `spartan` or `labelfusion` which provide docker environments with director pre-installed.

AlameluThinnappan, 2018-04-08:
Thank you!
I cleaned the existing build, built from source again with -DUSE_LCM=ON . It works now.

closed 2018-04-08 by AlameluThinnappan