RobotLocomotion/director #607: error using `start ik` button in kuka director app

manuelli, 2018-06-19:
When you hit `start ik` you get the following error

```
>>> Traceback (most recent call last):
  File "/home/manuelli/spartan/build/install/lib/python2.7/site-packages/director/teleoppanel.py", line 1004, in startIk
    om.removeFromObjectModel('reach goal')
  File "/home/manuelli/spartan/build/install/lib/python2.7/site-packages/director/objectmodel.py", line 530, in removeFromObjectModel
    _t.removeFromObjectModel(obj)
  File "/home/manuelli/spartan/build/install/lib/python2.7/site-packages/director/objectmodel.py", line 337, in removeFromObjectModel
    self._removeItemFromObjectModel(self._getItemForObject(obj))
  File "/home/manuelli/spartan/build/install/lib/python2.7/site-packages/director/objectmodel.py", line 241, in _getItemForObject
    return self._objectToItem[obj]
KeyError: 'reach goal'
```

it is due to trying to remove `reach goal` when it does not exist. I think an easy solution is to check if `reach goal` exists, and remove it if it does. Alternatively we can just use `vis.updateFrame` instead of `vis.showFrame` [here](https://github.com/RobotLocomotion/director/blob/master/src/python/director/teleoppanel.py#L1005) and drop line [1004](https://github.com/RobotLocomotion/director/blob/master/src/python/director/teleoppanel.py#L1004)

patmarion, 2018-06-20:
I think that the line of code is wrong:

```
om.removeFromObjectModel('reach goal')
```

Previously it must have silently failed and did nothing.  A code update a couple months ago refactored `objectmodel.py` and so now it actually catches the error.

The line of code in `teleoppanel.py` should be updated to:

```
om.removeFromObjectModel(om.findObjectByName('reach goal'))
```

patmarion, 2018-06-20:
I agree that it could also use `updateFrame` instead, but I think explicitly removing it and then showing a new one might be better for small reasons.

patmarion, 2018-06-20:
it looks like this line was clearing the whole parent folder, so removing the "reach goal" object probably isn't required at all:

https://github.com/RobotLocomotion/director/blob/master/src/python/director/teleoppanel.py#L643

PR #608 (2018-06-20): remove unnecessary line causing bug
fixes #607
closed 2018-06-20 by patmarion