RobotLocomotion/director #623: FYI: ROS-compatible version of Director exists

mauricefallon, 2019-05-13:
This is a note to users of Director. We (Oxford Dynamic Robot Systems) have adapted Director to natively compile within the ROS ecosystem:
https://github.com/ori-drs/director

PCL, OpenCV, VTK, Qt and other dependencies are gotten from ROS-land. Recently we updated to Ubuntu 18.04. LCM dependency has been removed.

We use it to operate our ANYmal robot instead of Rviz.

patmarion, 2019-05-14:
Thanks for the notice!  I will take a look.

I appreciate the need for a fork so that you can merge PRs as needed.  I wonder if you see the new work eventually coming back to Director `master` branch, or if the fork will become custom tailored to your project?

patmarion, 2019-05-14:
By the way, in case it is useful to you, there is a little more ROS code that I wrote that landed in `director/master` but probably after you had forked.  The code is here:

https://github.com/RobotLocomotion/director/commit/53ce0fbbc830d0261c83f7282758e52d7bdfa08f#diff-ffb2d4bdee542b713bd971ea3676c6db

patmarion, 2019-05-14:
The code in that commit shows how to create new director libraries, including python wrapping, in the `catkin_ws` subdir, without needing to add ROS code to the core director code in `src/app`

mauricefallon, 2019-05-15:
Hi Pat,
I don't have any plan to merge back to master right now. That said the folder structure is unchanged and the ROS changes are kept in a separate module - to keep it from infecting the core of director:
https://github.com/ori-drs/vtk_ros
Those are python-wrapped and loaded at runtime.

This vtk_ros subscribes/provides point clouds, images and depth images:
https://www.dropbox.com/s/o3fxxifj73v38ie/rough_terrain.mp4?dl=0
But work is needed to make this general and reusable.

A version of the DRC-era Director would compile without a ROS dependency at present.

We have an add-on repo which adapts the core Director app with UI components specific to ANYmal.
Most of our development is outside of the director repo and in that repo.

Basically, the question is "who would use this variant of Director?".
For us, Director must have a straightforward catkin build environment - cross platform compatibility is not important to us.