RobotLocomotion/director #628: ImportError:No module named PythonQt

Jerryxiaoyu, 2019-10-17:
I downloaded the latest Drake precompiled release and tried to run the following code:
```
 from director.consoleapp import ConsoleApp

app = ConsoleApp()
view = app.createView()
view.showMaximized()
app.start()

```
but got the error `ImportError: No module named PythonQt`
Is there something that I need to add to PYTHONPATH ?

patmarion, 2019-10-17:
Hi @Jerryxiaoyu,

The director package comes with a binary named `directorPython`.  Try using this instead of `python` to launch the program.   `PythonQt` is not a standard python module that can be found in the python path, it has to be initialized from C++.  The `directorPython` command line program is equivalent to `python` but it manages the initialization of Qt and PythonQt.

Jerryxiaoyu, 2019-10-18:
Great! I complied the director package and used `directorPython` to run the program. It works well.
Thanks a lot! @patmarion

patmarion, 2019-10-18:
Great, glad it worked!

I have a change that I've been experimenting with that removes the requirement of using `directorPython`.  It might land something like this in the future after a little more testing.

https://github.com/RobotLocomotion/director/pull/629

cc @EricCousineau-TRI