RobotLocomotion/director #630: Unable to run drake-visualizer after successful build

edrumwri, 2020-01-05:
I built with the default cmake options.

I pointed my PYTHONPATH to $DIRECTOR_HOME/src/python (not sure if this is the recommended workflow)

Running drake-visualizer gave me the following error message:

```
./build/install/bin/drake-visualizer 
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/drum/director/src/python/director/drakevisualizerapp.py", line 3, in <module>
    from director import mainwindowapp
  File "/home/drum/director/src/python/director/mainwindowapp.py", line 3, in <module>
    from director import consoleapp
  File "/home/drum/director/src/python/director/consoleapp.py", line 6, in <module>
    from director import applogic
  File "/home/drum/director/src/python/director/applogic.py", line 5, in <module>
    import director.vtkAll as vtk
  File "/home/drum/director/src/python/director/vtkAll.py", line 2, in <module>
    from director.vtkDRCFiltersPython import *
ModuleNotFoundError: No module named 'director.vtkDRCFiltersPython'
```

patmarion, 2020-01-05:
Hi @edrumwri you shouldn't set `PYTHONPATH`.  Director is composed of `.py` and `.so` files and they are all copied into the install tree under `build/install/lib/python3.6` (or whatever version of python was used to compile.  The `drake-visualizer` program will automatically search this path.  If you set `PYTHONPATH` then it finds `director` from the source tree which has the `.py` files but not the `.so` files.

edrumwri, 2020-01-05:
Hmm.. If I don't do that, then it has a different complaint:
`ModuleNotFoundError: No module named 'director'`

After building, should I be doing a `make install` somewhere?

This is all on a fairly fresh ubuntu 18 install, btw.

patmarion, 2020-01-05:
Director is tested in ubuntu 18, it should work.  It uses a "superbuild" which builds and installs all dependencies.  Running from the build/install dir should be all that is required, no extra steps.  Is there any more output, maybe printed before the `ModuleNotFoundError`?  If you look under build/install/lib/python3.6/site-packages do you see the director python files?

edrumwri, 2020-01-05:
No, that is the only output.

The files in build/install/lib/python3.6/site-packages (there is also a build/install/lib/python3.7/site-packages, which seems to have an identical list of files) follows:

actionhandlers.py           doordemo.py              lcmcollections.py       pointpicker.py             takktilevis.py
affordanceitems.py          drakevisualizerapp.py    lcmframe.py             pointselector.py           tasklaunchpanel.py
affordancemanager.py        drakevisualizer.py       lcmgl.py                polarisplatformplanner.py  taskrunner.py
affordancepanel.py          drcargs.py               lcmloggerwidget.py      profiler.py                tasks
affordanceupdater.py        drilldemo.py             lcmlogplayer.py         propertyanimation.py       tdx.py
affordanceurdf.py           drivingplanner.py        lcmobjectcollection.py  propertyset.py             teleoppanel.py
affordanceviewbehaviors.py  egressplanner.py         lcmoctomap.py           pydrakeik.py               terrainitem.py
applogic.py                 fieldcontainer.py        lcmspy.py               qtutils.py                 terrain.py
appsettings.py              filterUtils.py           lcmUtils.py             raycastdriver.py           terraintask.py
assimp2vtk.py               flags.py                 mainwindowapp.py        robotlinkselector.py       thirdparty
asynctaskqueue.py           footstepsdriverpanel.py  mapsregistrar.py        robotplanlistener.py       timercallback.py
atlasdriverpanel.py         footstepsdriver.py       matlabik.py             robotposegui.py            trackers.py
atlasdriver.py              frameupdater.py          matlab.py               robotstate.py              transformUtils.py
bihandeddemo.py             framevisualization.py    measurementpanel.py     robotsystem.py             treeviewer.py
blackoutmonitor.py          gamepad.py               meshmanager.py          roboturdf.py               uipanel.py
botspy.py                   geometryencoder.py       midi.py                 robotviewbehaviors.py      undoredo.py
callbacks.py                handcontrolpanel.py      motionplanningpanel.py  sceneloader.py             utime.py
camerabookmarks.py          handdriver.py            multisensepanel.py      screengrabberpanel.py      uuidutil.py
cameracontrolpanel.py       ikconstraintencoder.py   navigationpanel.py      segmentationpanel.py       valueslider.py
cameracontrol.py            ikconstraints.py         objectmodel.py          segmentation.py            valvedemo.py
camerainteractor.py         ikparameters.py          opendatahandler.py      segmentationroutines.py    version.py
cameraview.py               ikplanner.py             openscope.py            sensordatarequestpanel.py  viconvisualizer.py
colorize.py                 imageviewapp.py          optitrackvisualizer.py  shallowCopy.py             viewbehaviors.py
componentgraph.py           imageview.py             otdfmodel.py            simpletimer.py             viewcolors.py
consoleapp.py               __init__.py              outputconsole.py        sitstandplanner.py         viewerclient.py
continuouswalkingdemo.py    ioUtils.py               packagepath.py          skybox.py                  vieweventfilter.py
copmonitor.py               irisdriver.py            perception.py           splinewidget.py            visualization.py
coursemodel.py              irisUtils.py             plannerPublisher.py     spreadsheet.py             vtkAll.py
debrisdemo.py               jointcontrol.py          planningutils.py        startup.py                 vtkDRCFiltersPython.so
debugVis.py                 jointpropagator.py       planplayback.py         statelistener.py           vtkNumpy.py
depthimageprovider.py       kinectlcm.py             playbackpanel.py        surprisetask.py            walkingtestdemo.py
depthscanner.py             korgnano.py              pointcloudlcm.py        switchplanner.py           wristforcetorquevisualizer.py