RobotLocomotion/director #632: Debugging python code run by startup

heuristicus, 2020-10-14:
I am trying to debug some python code which is run in a customised startup.py that I am using. I would like to use a debugger such as `pdb` to do this but have so far been unable to make it work.

I have tried adding

```
import pdb
pdb.set_trace()
```

To the startup  script, but while I see some output from pdb I am unable to type any commands. I think this is likely caused by stdin/stdout not being linked to the terminal because of the way that the python file is executed.

Usually I would attach to a python process directly but again because of how PythonQt runs the python code, it is all contained within the single director process.

Is there any way I can get a debugger to attach to the python code?

patmarion, 2020-10-14:
Hi,

If the main window is open and responsive then you can use the built in Python console (press F8 to open it) to enter text to interact with pdb.  If you have set the trace before the application has finished initializing then you will have to use the terminal.  You can restore the stdin with this code before calling `set_trace`:

```
import sys
sys.stdin = sys.__stdin__
```

heuristicus, 2020-10-14:
Thanks, that was what I needed.

closed 2020-10-14 by heuristicus